emuflight/EmuConfigurator #8: linux64 compile error

meoso, 2019-09-17:
```shell
$ yarn test
yarn run v1.17.3
$ karma start test/karma.conf.js
17 09 2019 11:03:24.135:WARN [filelist]: All files matched by "/home/USER/Downloads/EmuConf/EmuConfigurator/node_modules/chai/chai.js" were excluded or matched by prior matchers.
17 09 2019 11:03:24.212:INFO [karma-server]: Karma v4.2.0 server started at http://0.0.0.0:9876/
17 09 2019 11:03:24.213:INFO [launcher]: Launching browsers ChromeHeadlessNoSandbox with concurrency unlimited
17 09 2019 11:03:24.220:INFO [launcher]: Starting browser ChromeHeadless
17 09 2019 11:03:25.216:INFO [HeadlessChrome 77.0.3865 (Linux 0.0.0)]: Connected on socket 1VhSCQFP_S7gVpo7AAAA with id 25317085
HeadlessChrome 77.0.3865 (Linux 0.0.0): Executed 13 of 13 SUCCESS (0.079 secs / 0.05 secs)
TOTAL: 13 SUCCESS
Done in 2.13s.
```
```shell
$ yarn gulp release --linux64 
yarn run v1.17.3
$ gulp release --linux64
[ '/usr/bin/node',
  '/home/USER/Downloads/EmuConf/EmuConfigurator/node_modules/.bin/gulp',
  'release',
  '--linux64' ]
Building for platform(s): linux64.
[11:02:10] Using gulpfile ~/Downloads/EmuConf/EmuConfigurator/gulpfile.js
[11:02:10] Starting 'release'...
[11:02:10] Starting 'clean_release'...
[11:02:10] Starting 'clean_apps'...
[11:02:10] Starting 'clean_dist'...
[11:02:10] Finished 'clean_release' after 17 ms
[11:02:10] Finished 'clean_apps' after 17 ms
[11:02:10] Finished 'clean_dist' after 183 ms
[11:02:10] Starting 'dist_src'...
[11:02:10] Finished 'dist_src' after 319 ms
[11:02:10] Starting 'dist_changelog'...
[11:02:10] Finished 'dist_changelog' after 2.61 ms
[11:02:10] Starting 'dist_yarn'...
[1/4] Resolving packages...
[2/4] Fetching packages...
info fsevents@1.2.9: The platform "linux" is incompatible with this module.
info "fsevents@1.2.9" is an optional dependency and failed compatibility check. Excluding it from installation.
info fsevents@2.0.7: The platform "linux" is incompatible with this module.
info "fsevents@2.0.7" is an optional dependency and failed compatibility check. Excluding it from installation.
[3/4] Linking dependencies...
[4/4] Building fresh packages...
success Saved lockfile.
[11:02:12] Finished 'dist_yarn' after 1.87 s
[11:02:12] Starting 'dist_locale'...
[11:02:12] Finished 'dist_locale' after 48 ms
[11:02:12] Starting 'dist_libraries'...
[11:02:12] Finished 'dist_libraries' after 23 ms
[11:02:12] Starting 'dist_resources'...
[11:02:12] Finished 'dist_resources' after 83 ms
[11:02:12] Starting 'getHash'...
[11:02:12] 90a16c2 
[11:02:12] Finished 'getHash' after 11 ms
[11:02:12] Starting 'writeChangesetId'...
[11:02:12] Finished 'writeChangesetId' after 15 ms
[11:02:12] Starting 'apps'...
Error building NW apps: Error: write EPROTO 139875219755520:error:141A318A:SSL routines:tls_process_ske_dhe:dh key too small:../ssl/statem/statem_clnt.c:2156:

[11:02:12] The following tasks did not complete: release, <parallel>, <series>, apps
[11:02:12] Did you forget to signal async completion?
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```

I know the ZIP file is available, but i don't want that. i want to compile a .deb to "install".
I am able to compile the old `ButterFlight/butterflight-configurator`.

Thank you

loutwice, 2019-09-17:
hello @meoso - i just tried yarn gulp release --linux64 and worked for me  , can you try yarn gulp apps --linux64 or even yarn start to see if it launch ?

loutwice, 2019-09-17:
seems like https://stackoverflow.com/questions/52478069/node-fetch-disable-ssl-verification?noredirect=1&lq=1

meoso, 2019-09-18:
fail on the two yarn commands, i'm follwing the lead you sent now... just unclear if it's an openssl setting or something elsewhere.  i will persist.

meoso, 2019-09-18:
woot! https://www.akeyes.co.uk/blog/debian_buster_first_install
solved: modify `/etc/ssl/openssl.cnf`
remove `CipherString = DEFAULT@SECLEVEL=2`
replace with `CipherString = DEFAULT@SECLEVEL=1`

closed 2019-09-18 by meoso