emuflight/EmuConfigurator #17: Mac OS damaged file error

flyandride, 2019-10-05:
I've tried downloading the Mac version several times.  I've gotten a very slow download that stalled and when it did finish I got a damaged file error.  

Downloading the Windows version also stalls.

besser, 2019-10-06:
That's a same issue that occour with betaflight configurator. If you clone this repo and compile your self the configurator run with no problem.

[https://imgur.com/a/VA8QQrt](url)

loutwice, 2019-10-06:
github sometime can mess with the download i will advise you to use a download manager as jdownloader for example   . let me know @flyandride

soulinux, 2019-10-09:
Which version works on Mojave 10.14.6?

loutwice, 2019-10-09:
0.2.0 should work - what is your error @soulinux ?

soulinux, 2019-10-12:
Thanks @loutwice,  problem solved with jdownloader

gretel, 2019-10-13:
for now i have hard time see any actual connection using a third party downloader.
am very suspicious on that being **not** a local issue, like stream-changing firewalls, antivirii, or alike.
personally, i can reproduce this issue both ways, not using anything like third party http client.

SupraDolph, 2019-10-19:
jdownloader also works for me. Thanks.

gretel, 2019-10-19:
we need to know what `jdownloader` changes. it's a file transferred via `http` there can't be no magic :)

SupraDolph, 2019-10-19:
I have no idea why that happen. It's the first time I encountered this kind of issue. I have also tried with the new version of Betaflight configurator. The same issue was happened.

gretel, 2019-10-19:
@SupraDolph we are working on providing signed apps for `osx` as soon as possible.

loutwice, 2020-01-01:
i think this one is solved by now? let me now

closed 2020-03-12 by gretel