emuflight/EmuConfigurator #21: Profile Copy is not working

N4V1G4T0R, 2019-10-28:
When trying to copy a Profile in the PID Tuning Tab, the settings are not copied to the target profile.

To reproduce, go into the PID Tuning Tab. Choose the profile you want to copy. Klick on "Copy profile values". Choose a Profile, where you want to copy the current settings to. (Press Save). Switch to the profile you chose to copy the settings to. You will see, that the copy did not work.
The same procedure works for Rateprofiles though.

Configurator is emuflight 0.2.0.

Quick-Flash, 2019-11-06:
This should be fixed with newer versions of the gui and versions of emuflight.

https://bintray.com/emuflight

You can easily try development versions of the software at that link!

loutwice, 2019-11-06:
actually testing on 0.2.9 on windows version - seems to works - will test more in depth

nerdCopter, 2020-01-11:
Only copying on from 1 to 2 seems to work. no other variations work. (tested 0.2.12)

ghden, 2020-06-28:
Same issue with latest 0.3.2 configurator, Profiles works copying from current to current + 1, you can copy from 1 to 2, but not for 1 to 3, and you can copy fine from 2 to 3.. Exact same error with rates profile..

nerdCopter, 2020-07-01:
Notes for @DEVS:
* i've looked at the code again.  the copy process seems to be part of the firmware MSP, not a process of the configurator.  The configurator just tells MSP the source and destination.  it uses the exact code for both pid-profile or rate-profile.
* rateprofile seems to be fully working as of test version 0.3.3 (and i recall testing in the past which worked also)
* i think this relates directly to #22  which also seems to be a firmware, non-configurator issue.

nerdCopter, 2020-07-01:
i'll keep it open as reminder for devs