emuflight/EmuConfigurator #22: Hangups switching from "PID Tuning Tab"

N4V1G4T0R, 2019-11-03:
I experience occasional hangups of the configurator when switching from the "PID Tuning Tab" to another tab. The configurator just keeps showing the PID Tuning Tab. I could not figure out the pattern of the hangup. It just sometimes occurs for no real reason.

Workaround: disconnect and reconnect.

gretel, 2019-11-08:
@N4V1G4T0R i'm also experience this type of bug, but the author can not reproduce it. i am sorry.

valeratlt80, 2019-11-12:
У меня в режиме настроек через OSD коптер тоже зависает иногда причем намертво даже не переключает полётные режимы тумблерами  помогает только жесткий перезапуск

loutwice, 2020-01-01:
this one is working now

nerdCopter, 2020-02-24:
i get this too, still. i think it's a firmware issue.

gretel, 2020-02-24:
ya me too.

gretel, 2020-03-13:
@emuflight/emuteam i am still getting this. it is annoying. please stop implementing new features but squish these ancient bugs instead. may be less fun but it is more important.

nerdCopter, 2020-05-23:
* specific issue source found: if `*.json` files do not exist, gui will crash (`pid_tuning.js`), but not exit.
* logical solution would be check for file existing before attempting to populate drop-down.

commit 217034fb (2020-05-26): fix failing presets for non-existent internet/json (#201)

* fixes #22
* preset  are stored in system temp folder; therefore, if temp  are non-existent, pid-tab would fail.
* solution is if  variable empty, do not populate presets list, but instead set it empty.
closed 2020-05-26 by nerdCopter