emuflight/EmuConfigurator #34: Aux 1 arming not working on kakute HD F7 and Emuflight 2.0

Ayougo, 2019-12-24:
Using 2.12 configurator and 2.0 Emuflight firmeware with the Kakute HD F7 fork (since the current release does not have the Kakute HD F7 option)... The "ARM" tab shows my radio commands and they correspond with the channel map BUT the color of the "ARM" mode doesnt change and I cannot arm my quad (default settings). The arm error messages are MSP and ARM switch (when enabled).

nerdCopter, 2020-01-11:
The configurator's ARM will never change color because it will technically not arm when plugged to usb. (i thought the exact same thing, opened an issue, then was told this).

Does the quadcopter arm on it's own?

gretel, 2020-03-13:
nerdcopter is correct. no issue.

closed 2020-03-13 by gretel