emuflight/EmuConfigurator #49: PID Tab is Blank.

NeilGaede, 2020-02-15:
Everything else works, including CLI that shows all the tasks are running, but when I click on the PID tab everything is blank, not just zeros, but blank. I can toggle values and change them, but there is nothing in the boxes under presets or anything.

NeilGaede, 2020-02-15:
Configurator 0.2.12 win64.

NeilGaede, 2020-02-15:
Tried configurator 2.11 and same problem.

NeilGaede, 2020-02-15:
Configuration version 2.0 works fine.

nerdCopter, 2020-02-24:
this happens on occasion. reboot/reinstall configurator, but also see 
https://github.com/emuflight/EmuFlight/issues/67#issuecomment-590365556
i.e. uninstall old versions & delete installation folder before installing new versions.

Canaill51, 2020-03-02:
I did have experienced same issue when was offline. When connecting to any Internet connection, issue was fixed. I am wondering if there is not a relation between presets updates and this concern? As presets are impacting PID and Filters... @nerdCopter ?

mdibe, 2020-03-07:
I can confirm the same issue happens to me when offline. Usually outside when flying at the field.
As soon as I get home and connect to the wifi the issue is solved.

nerdCopter, 2020-03-08:
now everything makes sense, thank you very much.  it's looking for the presets, which are online. 
one of us will look into a solution.

gretel, 2020-03-13:
@emuflight/emuteam there should be timeouts and error handling in place for all non-local transactions like http requests.

Mr-Buff, 2020-05-08:
Me too.. 
On the PID-config everything is blank..and after get in to PID-config GUI get stuck...
try to connect internet but doesn't work.
reinstall is work again in first connect and after that blank again.
use on Mac OS 10.14 Mojave.

AndreySemjonov, 2020-05-08:
What GUI version ?

gretel, 2020-05-08:
it's because the threads hang when there is no internet connection, or firewall blocking the requests. code needs to timeout and fail properly.

AndreySemjonov, 2020-05-08:
Its working for when no internet connection. Maybe it hang when there is connection, but can't download (or connection slow).

gretel, 2020-05-08:
it does not maybe hang but for sure.

AndreySemjonov, 2020-05-08:
I not talking about does it hang or not :) I try to replicate this issue and I can't. If I turn off internet connection it is working for me.

gretel, 2020-05-08:
the reporter mentioned a different operating system. you gonna say "i have windows only" now and keep this going on forever :)

AndreySemjonov, 2020-05-08:
Please look 1st report it tells about windows. To fix something I need to replicate it to check does it help or not. If you know how you can tell.

gretel, 2020-05-08:
cause i read the code you published. sorry for disturbance. i'm kinda like off here.

AndreySemjonov, 2020-05-08:
Your comment was really helpful thanks

PR #201 (2020-05-26): fix crashing pid-tab for failing non-existent presets/internet
* fixes #49 
* presets `.json` are stored in system temp folder; therefore, if temp `.json` are non-existent, pid-tab would fail.
* solution is if `presetJson` variable is empty, then do not populate presets list, but instead set it empty.

[image]
closed 2020-05-26 by nerdCopter