emuflight/EmuConfigurator #65: Mac OSX firmware flashing fail

mdibe, 2020-03-07:
Flashing a firmware on Mac OS seems to only work on 0.2.0 and previous.
All versions released after (0.2.11, 0.2.12, 0.2.12-RC2) are not able to flash the firmware:
The flight controller goes into DFU mode but then nothing happens.
0.2.0 seems to work fine even flashing the latest firmware.

mdibe, 2020-03-08:
Digging a little bit more it seems like the issue comes from the nwjs file.
0.2.0 uses 0.36.2-sdk which works fine.
0.2.11-12 use 0.42.2-sdk which is causing the issue.
I have found that the issue begins with version 0.37.1-sdk. So in the meantime I'm using configurator 0.2.13 with 0.37.0-sdk and seems to work fine.
I hope this could help to fix the problem.

gretel, 2020-03-12:
@mdibe thanks for the report! i have been flashing lots using the recent releases which have been build using `0.42.2` so i'm bit confused.

mdibe, 2020-03-12:
@gretel thanks. It might be that my MacOS High Sierra is too outdated somehow.

gretel, 2020-03-12:
@mdibe think same. sorry we can not support older versions of macos at the moment.
besides, it's possible to compile the configurator from source code, and change the definition accordingly at https://github.com/emuflight/EmuConfigurator/blob/a2c45a3c50b8825ec2c6aac6809053baf9596047/gulpfile.js#L43
you need `node10` and `yarn` installed:
```
yarn install
yarn gulp apps --osx64
```

PR #200 (2020-05-31): Fix OSX Flashing
* reportedly fixes flashing on MacOS version that previously failed
* update to nw-Builder dependency and gui fixes (workarounds) associated with such
* fixes #65. also reference #16, #51
closed 2020-03-12 by gretel