emuflight/EmuConfigurator #197: Beeping on MATEKF411RX (Tinyhawk S)

invadermike, 2020-05-20:
I tried to install it several times on my tinyhawk. And once I plug the battery, the song doesn't finish and it starts to beep forever.

Really look forward to try it :)

nerdCopter, 2020-05-21:
@invadermike 
* no reports of fails form others.  only reports of successes.
* please reflash 0.3.1 with full chip erase. (use configurator 0.3.2).
* do not use any diffs from other flightware.

please report any additional findings, thank you.

invadermike, 2020-05-21:
Think I found it, D16 works on betaflight but not on emuflight. But it
works with D8 so it's fine. Also the battery voltage needed to be higher.
But I just don't understand why the arming song doesn't finish and make the
beep beep beep sound like there was an error.

Anyway, it works now. Thanks !

closed 2022-07-22 by nerdCopter