emuflight/EmuConfigurator #207: CLRacing / F7 v2.1 / USER1 Mode missing

JohnAlfaJango, 2020-05-26:
I am using the following FC and cannot get the video to work properly.  On the Configurator 0.3.2, I am using the following version: # EmuFlight / CLRACINGF7 (CLR7) 0.3.1 May 19 2020 / 04:35:57 (9366544) MSP API: 1.47.  I have it configured and the quad will arm properly; however, I cannot get the video to get power from the FC using the USER1 mode.  Here is the DIFF ALL:
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff all

# version
# EmuFlight / CLRACINGF7 (CLR7) 0.3.1 May 19 2020 / 04:35:57 (9366544) MSP API: 1.47

board_name 
manufacturer_id 
mcu_id 002d00203338510b39393832
signature 

# reset configuration to default settings
defaults nosave

# name
name EMU CLRF7

# resources

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature OSD

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 32 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 4 900 1200 0 0
aux 1 27 3 1775 2100 0 0
aux 2 13 3 900 1200 0 0
aux 3 19 1 1300 1700 0 0
aux 4 35 1 1825 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_hz_roll = 130
set gyro_lowpass_hz_pitch = 130
set gyro_lowpass_hz_yaw = 130
set imuf_roll_q = 5500
set imuf_pitch_q = 5500
set imuf_yaw_q = 5500
set imuf_sharpness = 4000
set acc_calibration = -69,27,-18
set rssi_channel = 16
set fpv_mix_degrees = 30
set serialrx_provider = SBUS
set min_throttle = 1070
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set vbat_scale = 160
set beeper_dshot_beacon_tone = 5
set yaw_motors_reversed = ON
set tlm_halfduplex = OFF
set osd_units = IMPERIAL
set osd_warn_arming_disable = ON
set osd_warn_batt_not_full = ON
set osd_warn_batt_warning = ON
set osd_warn_batt_critical = ON
set osd_warn_visual_beeper = ON
set osd_warn_crash_flip = ON
set osd_warn_esc_fail = ON
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2092
set osd_rssi_pos = 2061
set osd_flymode_pos = 2368
set osd_throttle_pos = 2082
set osd_vtx_channel_pos = 2420
set osd_crosshairs_pos = 2285
set osd_ah_sbar_pos = 234
set osd_ah_pos = 234
set osd_mah_drawn_pos = 2071
set osd_craft_name_pos = 2440
set osd_altitude_pos = 2102
set osd_warnings_pos = 449
set osd_disarmed_pos = 2400
set osd_core_temp_pos = 506
set vtx_band = 1
set vtx_channel = 4
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5805
set vcd_video_system = PAL

# profile
profile 0

set dterm_lowpass_type = BIQUAD
set dterm_lowpass_hz_roll = 140
set dterm_lowpass_hz_pitch = 140
set dterm_lowpass_hz_yaw = 140
set dterm_lowpass2_hz_roll = 0
set dterm_lowpass2_hz_pitch = 0
set dterm_lowpass2_hz_yaw = 0
set witchcraft_roll = 0
set witchcraft_pitch = 0
set spa_roll_p = 100
set spa_roll_i = 100
set spa_roll_d = 100
set spa_pitch_p = 100
set spa_pitch_i = 100
set spa_pitch_d = 100
set spa_yaw_p = 100
set spa_yaw_i = 100
set spa_yaw_d = 100
set feathered_pids = 0
set i_decay = 6
set emu_boost = 250
set emu_boost_yaw = 250
set emu_boost_limit = 50
set emu_boost_limit_yaw = 50
set p_pitch = 70
set i_pitch = 60
set d_pitch = 34
set p_roll = 74
set i_roll = 60
set d_roll = 34
set p_yaw = 90
set d_yaw = 12

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 192
set pitch_rc_rate = 192
set yaw_rc_rate = 192
set roll_expo = 72
set pitch_expo = 72
set yaw_expo = 77
set roll_srate = 77
set pitch_srate = 77
set yaw_srate = 77
set tpa_rate_p = 60
set tpa_rate_i = 100
set tpa_rate_d = 70
set tpa_breakpoint = 1650

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
Write your command here. Press Tab for AutoComplete.

nerdCopter, 2020-05-26:
@Quick-Flash is this due the user4 vs user1 swap, or does Kaio need to look at target?
Does GUI need change?

mrrodgers, 2020-05-30:
I'm having the same issue. using Zeezf7. Needs user 1 mode to turn power on vtx.

COGNZNT, 2020-06-02:
Im having the same issue, using Talon F7 Fusion. It also needs user 1 mode to turn power on to the vtx.

Kaiowarez, 2020-06-02:
Okay
For all you guys not having video
Check resource PINIO 1 <###>
 Make sure it's not none
If it's filled out
Do set pinio_box =39,255,255,255
That should assign your vtx switch to vtx pidmode in modes tab
Then adjust the slider for it to work
Either have it on on-arm or just have the slider encompass the whole range so it's always on
If it's not working or stuff is missing hit me up on discord emuflight I'll try to help you more there
Kaio Pryde

Kaiowarez, 2020-06-02:
The talonf7fusion I remade to have the aux setup already I'm just jot sure if it got merged in with the new configurator
Unfortunately I can't do the same for the clracing since the older boards use the same target but didn't have a vtx pid switch built in.. Meaning the vtx is always on on the older boards

orloborlo, 2021-01-13:
My CLRacing F7 v2.1 is having similar problem - can not power on VTX with Pit mode and neither User1
diff all
The board has a switch for Real Pit VX on/off...
Any ideas???

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.6 Jan  5 2021 / 19:08:42 (a4b6db1e7) MSP API: 1.43
# config: manufacturer_id: CLRA, board_name: CLRACINGF7, version: 294f73cd, date: 2020-11-05T19:01:48Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CLRACINGF7
manufacturer_id CLRA
mcu_id 002700353438511236383538
signature 

# feature
feature GPS
feature LED_STRIP

# map
map TAER1234

# serial
serial 1 2 115200 57600 0 115200

# led
led 0 0,0::F:11
led 1 1,0::G:0
led 2 2,0::L:0
led 3 3,0::A:0
led 4 4,0::A:0
led 5 5,0::L:0
led 6 6,0::S:0
led 7 7,0::FTW:12

# mode_color
mode_color 3 5 9

# aux
aux 0 0 0 925 1250 0 0
aux 1 1 1 900 1600 0 0
aux 2 2 1 1650 2075 0 0
aux 3 39 3 900 2075 0 0

# master
set gyro_to_use = FIRST
set acc_trim_roll = -2
set acc_calibration = 0,27,-6,1
set mag_hardware = NONE
set baro_hardware = NONE
set min_throttle = 1070
set use_unsynced_pwm = OFF
set motor_pwm_rate = 480
set vbat_scale = 160
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 3

# save configuration
save
#

Kaiowarez, 2021-01-13:
Check for resource pinio 1
get pinio and get pinio_box
Those are the lines that is needed for users to work

orloborlo, 2021-01-16:
pinio1 is PA14 = pin 49 on F722
i checked directly on the pin and i am getting 3.3 V when User1 is active and 0 volts when User1 is not active.
So it is all OK with the setup, betaflight - which is the bad news = the board is defective.
I followed VTX+ traces to a transistor that has Vbat on one pin....

Kaiowarez, 2021-01-16:
Thank you for the clarification., and sorry to hear you got a defective
board bro

closed 2021-01-16 by nerdCopter