emuflight/EmuConfigurator #209: Bind issue with FRSKY_X_LBT rx/tx

SolberLight, 2020-05-27:
Hi ! Hope you're doing great ? :)

I'm trying to bind my meteor65 using F411RX on the latest stable version of emuflight 0.3.1

on betaflight i can successfully bind using :
ACCST D16
CH1-8
RxNum 63

and pasting thoose lines into the cli of betaflight :
set rx_spi_protocol = FRSKY_X_LBT
set frsky_spi_tx_id = 87,158
set frsky_spi_offset = -36
set frsky_spi_bind_hop_data = 2,50,100,150,200,15,65,115,165,215,30,80,132,180,230,45,95,145,195,10,60,110,160,210,25,75,125,175,225,40,90,140,190,5,55,105,155,205,20,70,120,170,220,35,85,135,185,255,255,255
set frsky_x_rx_num = 63

to gather thoose informations i did :
bind_rx, then after something like 1 min, I checked that my frsky_spi_bind_hop_data array was full of numbers then enter save, and voilà

so, i tryed pasting them into emu like this:
set rx_spi_protocol = FRSKY_X
set frsky_spi_tx_id = 87,158
set frsky_spi_offset = -36
set frsky_spi_bind_hop_data = 2,50,100,150,200,15,65,115,165,215,30,80,132,180,230,45,95,145,195,10,60,110,160,210,25,75,125,175,225,40,90,140,190,5,55,105,155,205,20,70,120,170,220,35,85,135,185,255,255,255
set frsky_x_rx_num = 63

but it wont work, i tryed the bind command and letting the radio bind for something like 2 mins then save, but nothing happen, do you have any idea on what is happening ?

Thanks for your help !

javitechjkd, 2020-06-16:
I am having exactly the same issue, but i am not 100% sure that we are doing binding correct, maybe is not an issue, there are another way of doing the binding?

SolberLight, 2020-06-16:
Well I've tryed many of them, but none of them worked yet

nerdCopter, 2020-06-16:
we dont have `_LBT`, but have this report from a tester:

3dracingman: (meteor65)  my  betafpv  hx100se uses it and i do not have problem using  FRSY_X or FRSY_D

```
# get frsky_spi_bind_hop_data
frsky_spi_bind_hop_data = 72,11,26,38,52,46,63,17,32,14,71,4,57,41,7,20,6,36,21,55,18,49,54,25,8,24,51,56,67,73,53,13,62,5,31,40,10,58,45,69,50,37,70,48,60,29,68,0,0,0
Array length: 50
Default value: 0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0

# version
# EmuFlight / MATEKF411RX (M41R) 0.2.53 May 13 2020 / 01:25:13 (7589952) MSP API: 1.46
```

SolberLight, 2020-06-28:
what is the process of binding then ? because bip radio + press bnd on the fc doesnt work

javitechjkd, 2020-06-28:
Finally with cli and the command bind and the tx set tod D8

Javier Corrales López

closed 2022-07-22 by nerdCopter