emuflight/EmuConfigurator #213: Can't re-arm after disarm.

SlavaChrome, 2020-05-30:
Emuflight 0.3.1
Frsky RXSR
Fport

It was ok before I installed 0.3.1 today morning.

Once I disarm it not arming again until I re-insert battery.
Also after disarm in 30 seconds Taranis says: "Sensor lost" and that is it.

Also the most terrifying thing, that you can't beep or turn on "Turtle Mode".

SolberLight, 2020-05-30:
Can it be gyro angle related thing ? Have you tryed putting max val on gyro angle arm ?

SlavaChrome, 2020-06-01:
I don’t think so!

I just arm, to check that motors spinning, don’t even take from ground, disarm. And after disarm I can’t re-arm.

It’s just lost connection.

nerdCopter, 2020-06-02:
@SlavaChrome , can you provide a diff and possibly a BBL of this behavior?

nerdCopter, 2020-06-02:
more notes: 
```
Risko: it seems to be when i made changes in the dji osd, then i cant re-arm and need to re-plug battery. if i do a quick hover test for example though, or sometimes after crash, i can still re-arm
Risko: its not only can't re-arm... beeper gone too, for me
Risko: CRSF
```

nerdCopter, 2020-06-04:
@SlavaChrome , are you using DJI?  it's been confirmed that switching profiles via DJI will cause this issue.
Emu has not focused on any DJI compatibility as of yet.  Any DJI working items are strictly by chance.

SlavaChrome, 2020-06-05:
Nope, I wrote, I use Frsky RXSR via Fport.

SlavaChrome, 2020-06-05:
> @SlavaChrome , can you provide a diff and possibly a BBL of this behavior?

Excuse me, had too much work and now it’s raining. What kind of logging do you need?

nerdCopter, 2020-06-09:
no problem. the devs & testers were trying to reproduce, but only one could reproduce the issue with DJI.

i guess the issue source is unknown at this time.  maybe logs could show was the idea.

nerdCopter, 2022-07-22:
closing; no activity.
 Also should use EmuFlight Discord for Help and Support https://discord.gg/BWqgBg3

closed 2022-07-22 by nerdCopter