emuflight/EmuConfigurator #242: Can't get CLI working well on 0.3.3

TcheTcheFPV, 2020-07-15:
Hi,

I wan't to inform you about a bug I'm facing on the last 0.3.3 configurator :

While going into the CLI, the "Building autocomplete cache" always fails, and I can't get a "diff all" running correctly until the end: it stops randomly before the end. See attached screenshot.
This happen with both win32 and win64 builds, and with all my quadcopter, and I always uninstall and delete the "emuflight" folder before installing another version.

Rolling back to 0.3.2 solve the issue.

Can you help ?
Thank you

[image]
[image]

nerdCopter, 2020-07-15:
what version and build number of Windows?

TcheTcheFPV, 2020-07-15:
Hi @nerdCopter ,

See attached the windows version and build.
As said, I always delete the emuflight folder in Program Files.

On my side, it happens all the time, and I can't get a diff all running until the end.

[image]

TcheTcheFPV, 2020-07-15:
[image]

nerdCopter, 2020-07-16:
* @TcheTcheFPV man, i'm sorry i cannot find a reason for this. 
* With the configurator closed, please delete the folder `C:\Users\%USERNAME%\AppData\Local\emuflight-configurator` to remove any local settings. 
* If that fails to fix it, you may disable the autocomplete cache entirely, from the gear-menu.
[image]
* I've added `.zip` releases for both  0.3.2 and 0.3.3 as a convenience of not requiring installation.  You can unzip the contents and run the .exe directly.  if you want both releases, you can rename the extracted folders. I've wanted to do this awhile and finally figured out how, so we will continue to release .zip moving forward.
* I'll continue to investigate the issue.
* Please continue to feedback with your findings, thank you!

TcheTcheFPV, 2020-07-16:
Hi @nerdCopter,

I have downloaded zip file of 0.3.2 and 0.3.3 both win32 and win64. First thing: both win32 and win64 of the 0.3.2 version are not working. running the exe file ends with the following screenshot.

I have then tried both win32 and win64 from 0.3.3 version, disabling autocomplete and deleting `C:\Users\%USERNAME%\AppData\Local\emuflight-configurator` folder each time and this issue is still here : I can't get a `diff all` running until the end.

Someting else : when using the 0.3.3 version, it does not launch full screen, even if it was before closing. On 0.3.2, it "remember" that it was full screen.

[image]

nerdCopter, 2020-07-16:
* I verified 0.3.2 zip problem, i'll delete and try to rectify.
* Only delete local folder only once (not every time). this is where it remembers screen/size/position and other options.

* if you disable autocomplete cache, CLI i still a problem?

TcheTcheFPV, 2020-07-16:
Even when I don't delete the local folder, it does not remember the screen size.
Yes, even when I disable the autocomplete cache, the CLI still does not provide the full `diff all` output.

nerdCopter, 2020-07-16:
* the size/position thing is a windows specific issue. there is no code that specifies this behavior.
* i'll try to investigate the CLI, but i cannot reproduce it at the moment :(
* please upload a save-file from the `diff all` output so i can understand better

nerdCopter, 2020-07-16:
i fixed the 0.3.2 `.zip`

TcheTcheFPV, 2020-07-16:
Here is the requested files, and the screenshot related to the `diff all` command.
As you see on the screenshot, it stops like that, with a half of a command in the input window. 
Very strange...

[image]
[EMUF_cli_DUMPALL_tchetche_20200716_210208.txt](https://github.com/emuflight/EmuConfigurator/files/4933645/EMUF_cli_DUMPALL_tchetche_20200716_210208.txt)
[EMUF_cli_DIFFALLtchetche_20200716_210057.txt](https://github.com/emuflight/EmuConfigurator/files/4933646/EMUF_cli_DIFFALLtchetche_20200716_210057.txt)

nerdCopter, 2020-07-18:
@TcheTcheFPV , zero guarantees, but if you like to test this build and report back, then we thank you.
https://artprodcus3.artifacts.visualstudio.com/A6d8d651a-56f1-4641-87d1-4c6fd7323cb6/259f263d-b6c9-41d1-8087-4004562227fe/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL25lcmRDb3B0ZXJBbm9uL3Byb2plY3RJZC8yNTlmMjYzZC1iNmM5LTQxZDEtODA4Ny00MDA0NTYyMjI3ZmUvYnVpbGRJZC8yMTIvYXJ0aWZhY3ROYW1lL2VtdWZsaWdodC1jb25maWd1cmF0b3Itd2luZG93cw2/content?format=file&subPath=%2Femuflight-configurator_0.3.3_win64.zip

TcheTcheFPV, 2020-07-18:
Hi @nerdCopter thank you for trying helping me. Unfortunately there is no change with this build.

I have also noticed something : is it normal to get randomly some signs and command on top of the CLI window just before it build the autocomplete cache ? (see screenshot below).

[image]

nerdCopter, 2020-07-18:
i've seen random characters (typical serial connection stuff) plenty of times, but not necessarily a list of items like your screenshot.

nerdCopter, 2020-07-18:
@TcheTcheFPV , Can you test this one? if this still fails then it is either a third party dependency problem that we have no control over (`NW.js`) or a Windows build/driver/update issue.  i suspect the `NW.js` which is the "node-webkit" that contains the Chromium versions that the App is built upon. different versions have broken flashing on OSX and Windows7 (differently), and possibly now somehow USB in Windows. there are som nay versions where some break different OS's (https://dl.nwjs.io/).

If this does work, however, it means we may have to revert to older MSP.js code or somehow fix the code we used from BeF.

https://artprodcus3.artifacts.visualstudio.com/A6d8d651a-56f1-4641-87d1-4c6fd7323cb6/259f263d-b6c9-41d1-8087-4004562227fe/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL25lcmRDb3B0ZXJBbm9uL3Byb2plY3RJZC8yNTlmMjYzZC1iNmM5LTQxZDEtODA4Ny00MDA0NTYyMjI3ZmUvYnVpbGRJZC8yMTMvYXJ0aWZhY3ROYW1lL2VtdWZsaWdodC1jb25maWd1cmF0b3Itd2luZG93cw2/content?format=file&subPath=%2Femuflight-configurator_0.3.3_win64.zip

nerdCopter, 2020-07-18:
This is out master source code plus the latest `NW.js` 0.47.0 -- it would interesting to test/compare this build as well:
https://artprodcus3.artifacts.visualstudio.com/A6d8d651a-56f1-4641-87d1-4c6fd7323cb6/259f263d-b6c9-41d1-8087-4004562227fe/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL25lcmRDb3B0ZXJBbm9uL3Byb2plY3RJZC8yNTlmMjYzZC1iNmM5LTQxZDEtODA4Ny00MDA0NTYyMjI3ZmUvYnVpbGRJZC8yMTQvYXJ0aWZhY3ROYW1lL2VtdWZsaWdodC1jb25maWd1cmF0b3Itd2luZG93cw2/content?format=file&subPath=%2Femuflight-configurator_0.3.3_win64.zip

nerdCopter, 2020-07-18:
@TcheTcheFPV , i was just told that on the ones that fail to complete the `dump all`, you can enter a few random commands or jibberish on the CLI input, pressing enter with each command until eventually the data will show/finish again.
Apparently this is only an issue on some Windows versions.  Somehow the buffer is incomplete or not dumping properly.  Can you verify this circumvents the problem?

TcheTcheFPV, 2020-07-19:
Hi @nerdCopter 

I've tried the build with the old version of MSP.js, and it does not solve the issue, the behavior remains the same.
In the meantime I have updated all my drivers: Zadig, CP210, and STM with no success.

So, I tried the workaround you suggested, and it works, so it might be something related to the buffer as you said ...

See :

[image]
1: enter `diff all` and hit enter.

[image]
2: it stops (seems to be always at the same moment)

[image]
3: Hit `enter `key: it delete the input window content

[image]
4: Enter something in the input window : here only one letter, and hit `enter`

[image]
5: the `diff all` continues until the end, but the `x` we enter before is also outputed after the `diff all` output

[image]
6: if we scroll up to the moment it has stopped, we can see that the half of the command that was in the input window and that we deleted by hitting enter, is anyway here, so we can `save to file` and get the full `diff all` backed up

[image]
7: the only thing we have to pay attention is that the command we entered before (the letter x) is also backed up at the end of the file.

nerdCopter, 2020-07-19:
* Thanks @TcheTcheFPV , none of us wrote that specific code (CLI/cache/buffer), so it may take awhile to analyze/fix.
* In the meanwhile i opened a github issue with `NW.js` which may lead to nothing. We are limited to NW.js 0.45.6 at this time. 0.46.0 and above do not list COM ports on Windows 7. (https://github.com/nwjs/nw.js/issues/7524)
* It's also very hard to determine fixes when i cannot reproduce the issue.  It seems running Windows VM's on Linux does not suffer from the same bug as Windows on bare-metal hardware.

nerdCopter, 2020-07-19:
Welcome to test another... i increased the `lineDelayMs: 15,` to `lineDelayMs: 30,` in source code.
https://artprodcus3.artifacts.visualstudio.com/A6d8d651a-56f1-4641-87d1-4c6fd7323cb6/259f263d-b6c9-41d1-8087-4004562227fe/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL25lcmRDb3B0ZXJBbm9uL3Byb2plY3RJZC8yNTlmMjYzZC1iNmM5LTQxZDEtODA4Ny00MDA0NTYyMjI3ZmUvYnVpbGRJZC8yMjAvYXJ0aWZhY3ROYW1lL2VtdWZsaWdodC1jb25maWd1cmF0b3Itd2luZG93cw2/content?format=file&subPath=%2Femuflight-configurator_0.3.3_win64.zip

TcheTcheFPV, 2020-07-20:
Thanks @nerdCopter for trying fixing the issue. Unfortunately this last one still hangs on the output.
Hope something will be fixed in a next windows update.

TcheTcheFPV, 2020-07-20:
I've just noticed that it happen also on 0.3.2 but rarely.
On 0.3.3 it's every time.
For both it's on the same computer.

nerdCopter, 2020-07-20:
@TcheTcheFPV , i was told this version works more like 0.3.2 ... i.e. it does not increase the buffer issue.
It uses `NW.js` version `0.42.6`. 0.3.2 used 0.42.
https://artprodcus3.artifacts.visualstudio.com/A6d8d651a-56f1-4641-87d1-4c6fd7323cb6/259f263d-b6c9-41d1-8087-4004562227fe/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL25lcmRDb3B0ZXJBbm9uL3Byb2plY3RJZC8yNTlmMjYzZC1iNmM5LTQxZDEtODA4Ny00MDA0NTYyMjI3ZmUvYnVpbGRJZC8yMjIvYXJ0aWZhY3ROYW1lL2VtdWZsaWdodC1jb25maWd1cmF0b3Itd2luZG93cw2/content?format=file&subPath=%2Femuflight-configurator_0.3.3_win64.zip

i also have a `NW.js` `0.43.6` for testing: 
https://artprodcus3.artifacts.visualstudio.com/A6d8d651a-56f1-4641-87d1-4c6fd7323cb6/259f263d-b6c9-41d1-8087-4004562227fe/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL25lcmRDb3B0ZXJBbm9uL3Byb2plY3RJZC8yNTlmMjYzZC1iNmM5LTQxZDEtODA4Ny00MDA0NTYyMjI3ZmUvYnVpbGRJZC8yMjQvYXJ0aWZhY3ROYW1lL2VtdWZsaWdodC1jb25maWd1cmF0b3Itd2luZG93cw2/content?format=file&subPath=%2Femuflight-configurator_0.3.3_win64.zip

These do not change the delay-milliseconds like the last, since it seemed to not help.

TcheTcheFPV, 2020-07-21:
@nerdCopter I got a good news : 

The first one you gave on your last message, the one with `NW.js` `0.42`,  is working ! 
But not the one with `NW.js` `0.43.6`.

So, 0.3.3 with `NW.js` `0.42` is working for me, and also the autocomplete cache buiding does not fail anymore (maybe it will sometimes as it did before).

How this will be handled in the future configurator releases ?

Thanks :)

nerdCopter, 2020-07-21:
* I suppose we have to revert to very old versions of `NW.js` :(
* It is odd that other testers report `0.43.6` worked as well.
* Using old versions of `NW.js` is essentially using old version of `Chromium` to build the app.

nerdCopter, 2020-12-15:
closing this issue. merged #250 Aug 21.
remaining profile changes seem to be firmware side.
#287 may assist.

closed 2020-12-15 by nerdCopter