emuflight/EmuConfigurator #247: Please Add default preset clearly in PID Tuning

simoccc, 2020-07-16:
Hi,

Pid Tuning TAB:
Preset menu at the bottom there are 2 profiles "default" one of them is ( correct me if I'm wrong ) the true default setting, the other is with slightly different LPF filters.

What I'm asking is to change the names to only default only for the one of the two that is exactly with the same numbers as defaults .

I hope that is clear, let me know if you need further explanation,

Best Regards,

Simone (Simofly Fpv )

nerdCopter, 2020-07-17:
There is a reason for such.
The lower dTerm LPF may be more safe for someone that has hot motors, and will be the default in the next release.
The one with higher dTerm LPF is what we think are "better" tuned dTerm LPF; However, some people have complained about hot motors. So as a solution, we will lower the defaults to what we had in 0.2.0 and allow the end-user to choose a preset.

nerdCopter, 2020-07-17:
I'm sure we would be open to better verbiage here:
[image]
[image]

simoccc, 2020-07-17:
Thanks for the answer, my request is just to have a clear indication if I want to revert back to defaults pids, to have a simple "default" preset just to have a clear indication, that's it 👍

nerdCopter, 2020-07-17:
* Thank you :)
* Modified verbiage for some affect: https://github.com/emuflight/emuflight-presets/pull/8
* PIDS and other feature values are the same for either of the two defaults. Only dTerm LPF are different which usually need tuning regardless (along with gyro LPFs).

closed 2020-07-17 by nerdCopter