emuflight/EmuConfigurator #285: Rate Profile also cause USB to hang

Max-Headroom, 2020-11-10:
Known issue: "Switching to PID Profile 2 or 3 (profile 1 or 2 in CLI) has been known to hang the USB connection, requiring unplug/replug."

This happens with changing Rate Profile as well. But very little.

Windows 10 Home edition
Configurator: 0.3.4

kaji331, 2020-11-15:
Also Remote tab hang in 0.3.4 configurator, Ubuntu 16.04/Windows 10

nerdCopter, 2020-12-03:
if you gys want to join the discord, i can send you link to a build that has a longer profile-change-timout, i have no proof it works, but have been testing it myself.  that said, i have fewer profile-hang with last release than prior, so i dont know if the code i'm offereing has any real affect or not.
just @nerdCopter  in discord and request it.
https://discord.com/invite/gdP9CwE

nerdCopter, 2022-07-22:
newer EmuConfig seem to resolve this for the most part. closing ticket.

closed 2022-07-22 by nerdCopter