emuflight/EmuConfigurator #309: macOS Big Sur crashes

truespin, 2021-01-29:
EmuConfigurator is currently very crashy on macOS Big Sur 11.x
BetaFlight Configurator has the same issue and have tracked/fixed it here, can this fix get pulled into EmuConfigurator?
https://github.com/betaflight/betaflight-configurator/issues/2274

MacBook Pro 2020 M1, macOS 11.1, EmuConfigurator 0.3.5

nerdCopter, 2021-01-29:
we dont even sign the code, so ...hmmm.. i'll look but unsure at the moment.
That command in the comments is a local machine command you might try on your own... :shrug:

truespin, 2021-01-29:
I tried that and got:
`objc[55146]: Class AMSupportURLConnectionDelegate is implemented in both ?? (0x1f5e038f0) and ?? (0x1178702b8). One of the two will be used. Which one is undefined.
objc[55146]: Class AMSupportURLSession is implemented in both ?? (0x1f5e03940) and ?? (0x117870308). One of the two will be used. Which one is undefined.`

nerdCopter, 2021-01-29:
@truespin , Please try this and let me know if it solves the issue, TY!
 https://drive.google.com/drive/folders/1bcPCEUpN1K9l3JF8CLI37TUigCppq-TA?usp=sharing

truespin, 2021-01-30:
Wow, that was quick, had a good play for 10 mins with my Mockula 6 on M1 MBP and no crashes!
Thanks, I really appreciate this :)

nerdCopter, 2021-02-10:
@truespin , has it continued to work without issue?

truespin, 2021-02-10:
I'd had no need to fire it up since!
But just tweaked a few settings on my Mockula6, visited each tab etc. and again zero crashes :)
Am happy for this to be closed if you are, should I do so or do you?

nerdCopter, 2021-02-10:
i'd like to leave it open for other pilots to see link until it's released.  would be nice to get more feedback from other pilots as well.     TY @truespin

krzaczek, 2021-02-14:
> @truespin , Please try this and let me know if it solves the issue, TY!
> https://drive.google.com/drive/folders/1bcPCEUpN1K9l3JF8CLI37TUigCppq-TA?usp=sharing

Seems to work just fine on a M1 Macbook pro

TXBDan, 2021-05-11:
Hi all, I'm having this issue with version 0.3.5. Where can I find the fixed version? I can't find any betas?

nerdCopter, 2021-05-11:
> Hi all, I'm having this issue with version 0.3.5 and the google drive link expired. Where can I find the fixed version? I can't find any betas?

https://github.com/emuflight/dev-master/releases/tag/cfg-179

nerdCopter, 2021-06-24:
released: https://github.com/emuflight/EmuConfigurator/releases/tag/0.3.6

closed 2021-06-24 by nerdCopter