emuflight/EmuConfigurator #332: Cli overflow issue with diff all and dump all

Max-Headroom, 2021-03-07:
Configurator: 0.3.5
Firmware: EMUF 0.3.3
Target: IFRC_IFLIGHT_F745_AIO (test target)
Windows 10

When I do a diff all or dump all, the data stops at about 3/4 of the way down. When I scroll back up in history, then the rest of the data is at the top.

I'm not sure if this is a firmware issue, as it looks like it is the way the configurator shows the data it receives.

My diff as is:
= 70
set yaw_srate = 70
set das_yaw_with_roll_input = 0
set das_roll_with_yaw_input = 0
set roll_pitch_mag_expo = 0
set rate_center_sensitivity = 100
set rate_end_sensitivity = 100
set rate_center_correction = 10
set rate_end_correction = 10
set rate_center_weight = 0
set rate_end_weight = 0
set tpa_rate_p = 75
set tpa_rate_i = 125
set tpa_rate_d = 65
set tpa_breakpoint = 1600
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set vbat_comp_type = OFF
set vbat_comp_ref = 37

# rateprofile
rateprofile 5

set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set das_yaw_with_roll_input = 0
set das_roll_with_yaw_input = 0
set roll_pitch_mag_expo = 0
set rate_center_sensitivity = 100
set rate_end_sensitivity = 100
set rate_center_correction = 10
set rate_end_correction = 10
set rate_center_weight = 0
set rate_end_weight = 0
set tpa_rate_p = 75
set tpa_rate_i = 125
set tpa_rate_d = 65
set tpa_breakpoint = 1600
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set vbat_comp_type = OFF
set vbat_comp_ref = 37

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# diff all

# version
# EmuFlight / IFRC_IFLIGHT_F745_AIO (IFRC) 0.3.3 Feb 14 2021 / 16:48:23 (ae6a0bd2a) MSP API: 1.50

board_name 
manufacturer_id 
mcu_id 003e003c3039510c34323433
signature 

# reset configuration to default settings
defaults nosave

# name
name Protek 35

# resources

# mixer

# servo

# servo mix

# feature
feature -AIRMODE
feature MOTOR_STOP
feature TELEMETRY
feature LED_STRIP
feature OSD

# beeper
beeper -GYRO_CALIBRATED
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 1 64 115200 57600 0 115200
serial 6 8192 115200 57600 0 115200

# led
led 0 0,0::L:0
led 1 1,0::L:0
led 2 2,0::L:0

# color

# mode_color
mode_color 5 0 0
mode_color 5 1 0
mode_color 5 2 0
mode_color 5 3 0
mode_color 5 4 0
mode_color 5 5 0

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 13 2 1675 2075 0 0
aux 3 28 7 1700 2100 0 0
aux 4 31 3 1700 2100 0 0
aux 5 35 5 1700 2100 0 0
aux 6 39 4 1700 2100 0 0

# adjrange
adjrange 0 0 6 900 1200 12 6 0 0
adjrange 1 0 6 1300 1625 12 6 0 0
adjrange 2 0 6 1700 2100 12 6 0 0

# rxrange

# vtx

# rxfail

# master
set imuf_roll_q = 8500
set imuf_pitch_q = 8500
set imuf_yaw_q = 8500
set imuf_sharpness = 1
set acc_calibration = 69,15,-9
set mag_hardware = NONE
set rssi_scale = 200
set serialrx_provider = FPORT
set serialrx_inverted = ON
set airmode_start_throttle_percent = 35
set serialrx_halfduplex = ON
set current_meter = ADC
set battery_meter = ADC
set yaw_motors_reversed = ON
set osd_warn_arming_disable = ON
set osd_warn_batt_not_full = ON
set osd_warn_batt_warning = ON
set osd_warn_batt_critical = ON
set osd_warn_visual_beeper = ON
set osd_warn_crash_flip = ON
set osd_warn_esc_fail = ON
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = ON
set osd_warn_dji = OFF
set osd_rssi_alarm = 35
set osd_rssi_pos = 2115
set osd_tim_2_pos = 2455
set osd_flymode_pos = 429
set osd_throttle_pos = 2423
set osd_vtx_channel_pos = 2082
set osd_current_pos = 2433
set osd_altitude_pos = 2103
set osd_avg_cell_voltage_pos = 2401
set vtx_channel = 3
set vtx_power = 3
set vtx_freq = 5780

# profile
profile 0

set dterm_lowpass_hz_roll = 100
set dterm_lowpass_hz_pitch = 100
set dterm_lowpass_hz_yaw = 100
set dterm_lowpass2_hz_roll = 250
set dterm_lowpass2_hz_pitch = 250
set dterm_lowpass2_hz_yaw = 250
set smart_dterm_smoothing_roll = 75
set smart_dterm_smoothing_pitch = 75
set smart_dterm_smoothing_yaw = 75
set witchcraft_roll = 6
set witchcraft_pitch

Max-Headroom, 2021-03-07:
Should have done this before reporting.
Just tested this with another quad with release firmware. and all is fine.

Before closing issue. Just for education. How does the data dump work?
I take it CLI (Command Line Interface) is a terminal where you communicate direct with the firmware. 
How is the data read when doing a dump? is it line by line, or a file or buffer that is read by the configurator?

nerdCopter, 2021-03-17:
Seems some people have similar issues, but not all FC/USB are created equal.
One workaround might be to reduce gyro/pid loops to something small (1k/1k), perform all configurations, then restore to desired loop frequencies before flying.

nerdCopter, 2021-03-17:
@Max-Headroom , What OS is this on Win7, Win10, OSX, Linux?

nerdCopter, 2021-03-17:
yes, loop frequencies _sometimes_ interfering with USB communication or Configurator serial/buffer. This fork is older and BF had had many improvements since.  i've seen pilots have similar problem.

nerdCopter, 2021-03-17:
Welcome to test this build for Windows 10 only: 
https://drive.google.com/drive/folders/1oQX0WORTUcRTU04GHpYXjKSpEUsgD22e?usp=sharing

Max-Headroom, 2021-03-18:
I am using that build for Windows 10.

But yes, changing loop frequency to 1KHz did fix it. Thanks! 
Hope this can be updated at some point. 
Not a major issue now, but at least we know what the problem is. Putting my microprocessor hat on, I can understand why.
Love tuning loops to fit in with everything doing its own thing, haha.

nerdCopter, 2021-03-19:
I think no effort will go into fixing this because EmuFlight is rebasing on BF 4.x. and a new UI will come with that as well.
But i'm very glad the loop freq helped. :)